andresarmento/modbus-esp8266 #3: Modbus serial

hallard, 2016-02-08:
Hi, 
Thanks a lot for this lib. I'm trying to use modbus serial on a ESP8266 but I'm not sure what to do, do I need to take ModbusSerial lib from modbus-arduino (there is none on esp8266 lib) and Modbus library from modbus-esp8266, is that correct ?

DIYDave, 2017-09-19:
Hi
Same here.
Works fin with Arduino Nano and HW-Serial. (brilliant Library!)
But with ESP8266 and HW-Serial I have Errors when compiling.
I'm a newbie, pleas help.

--------- Error message: -------------

Arduino: 1.8.3 (Windows 10), TD: 1.37, Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

H:\Users\Dave\Documents\Arduino\libraries\ModbusSerial\ModbusSerial.cpp: In member function 'bool ModbusSerial::config(HardwareSerial*, long int, u_int, int)':

H:\Users\Dave\Documents\Arduino\libraries\ModbusSerial\ModbusSerial.cpp:23:31: error: invalid conversion from 'u_int {aka unsigned int}' to 'SerialConfig' [-fpermissive]

     (*port).begin(baud, format);

                               ^

In file included from H:\Users\Dave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Arduino.h:245:0,

                 from H:\Users\Dave\Documents\Arduino\libraries\ModbusSerial\ModbusSerial.h:5,

                 from H:\Users\Dave\Documents\Arduino\libraries\ModbusSerial\ModbusSerial.cpp:5:

H:\Users\Dave\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/HardwareSerial.h:77:10: error:   initializing argument 2 of 'void HardwareSerial::begin(long unsigned int, SerialConfig)' [-fpermissive]

     void begin(unsigned long baud, SerialConfig config)

          ^

hallard, 2017-09-19:
I've got a fix for that ;-) 

in ModbusSerial.cpp
```cpp
#ifdef ESP8266
bool ModbusSerial::config(HardwareSerial* port, long baud,  SerialConfig format, int txPin) 
#else
bool ModbusSerial::config(HardwareSerial* port, long baud, u_int format, int txPin) 
#endif
```

in ModbusSerial.h
```cpp
        #ifdef ESP8266
        bool config(HardwareSerial* port, long baud,  SerialConfig format, int txPin) ;
        #else
	bool config(HardwareSerial* port, long baud, u_int format, int txPin=-1);
	#endif
````

DIYDave, 2017-09-22:
Thanks for your tip! 
I have solved the problem quick and dirty. I have made the setting directly in the Modbus Library. 
Now I will make it right in your way.

ayaser, 2021-12-29:
mucho gusto a todos.
Tengo el mismo problema que se describe mas arriba:  void begin(unsigned long baud, SerialConfig config)
Intente buscar el archivo ModbusSerial.h en mis carpetas, pero no lo encontre. Tal vez deba crear dicho archivo, o descargarlo con una libreria mas actualizada. saludos cordiales.

ayaser, 2021-12-30:
andre buenas noches. tengo problemas con la libreria ModbusSerial.h
Esta en el directorio habitual, pero Arduino Ide no la encuentra:
[image]
la imagen adjunta, es precisamente lo que me muestra Arduino IDE. Una de las librerias la muestra en rojo, que segun lo que he ido aprendiendo es que la vio, la encontro en el directorio de librerias, mientras que la que no encuentra esta en negro.
Alguna sugerencia de cual es el error y como resolverlo?
saludos cordiales.