andresarmento/modbus-esp8266 #4: Can not read   mb.Ists(INPUTS [7], 1);

isaacTN, 2016-02-14:
I'm having problems reading   mb.Ists(INPUTS [7], 1);  I modified your code.  

/*
  Modbus-Arduino Example - Test Holding Register (Modbus IP ESP8266)
  Configure Holding Register (offset 100) with initial value 0xABCD
  You can get or set this holding register
  Copyright by André Sarmento Barbosa
  http://github.com/andresarmento/modbus-arduino
*/

#include <ESP8266WiFi.h>
#include <Modbus.h>
#include <ModbusIP_ESP8266.h>

// Network Settings  Static IP
IPAddress ip(192,168,1,102);  
IPAddress gateway(192,168,1,1);
IPAddress subnet(255,255,255,0);

// Setup arrays addresses [0 .. 31] for addresses 0 to 31
const int INPUTS      [32]= {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31};
const int COILS       [32]= {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31};
const int INPUT_REG   [32]= {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31};
const int HOLDING_REG [32]= {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31};

int index1 = 0;
long timer0 = 0;

//ModbusIP object
ModbusIP mb;
  
void setup() {

  WiFi.config(ip, gateway, subnet);      
  Serial.begin(115200); 
  mb.config("PFPS-2.4G", "Samera#$1977");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

/****************************************************/ 
  // Address 1 on the PLC is Address 0 on the ESP8266  
  for (index1 = 0; index1 < 32; index1++)        
    {
      mb.addIsts(INPUTS [index1]);        
    }        
/****************************************************/
  // Address 1 on the PLC is Address 0 on the ESP8266      
  for (index1 = 0; index1 < 32; index1++)        
    {
      mb.addCoil(COILS [index1]);     
    }           
/****************************************************/
  // Address 1 on the PLC is Address 0 on the ESP8266  WORKING
  for (index1 = 0; index1 < 32; index1++)        
    {
      mb.addIreg(INPUT_REG [index1]);       
    }    
/****************************************************/ 
  // Address 1 on the PLC is Address 0 on the ESP8266  WORKING
  for (index1 = 0; index1 < 32; index1++)        
    {
      mb.addHreg(HOLDING_REG [index1]);      
    }
/****************************************************/
    
}
 
void loop() {
   //Call once inside loop() - all magic here
   mb.task();
      if (millis() > timer0 + 2000) {
       timer0 = millis();
       mb.Ists(INPUTS [7], 1);
         
      }             
  
}

ioteng, 2018-08-09:
Is there anyone who solved this problem?

ESP-O-MAT, 2020-04-14:
I think this is related to this issue:
https://github.com/andresarmento/modbus-arduino/issues/35