andresarmento/modbus-esp8266 #7: adding regs

h-key, 2016-06-02:
can you help, look at my project file and me what's wrong, pleas
[ESP12E_modified.zip](https://github.com/andresarmento/modbus-esp8266/files/294889/ESP12E_modified.zip)
e

h-key, 2016-06-04:
Here is  what's in ESP12E_modified.zip
 Let me know, What and where I've went wrong.
I looked at the examples and there is only one entry in each example, and its hard to see any pattern
and I'm trying but it doesn't seem to be working for me

/* Modbus
      Andresarmento's ESP8266 Modbus
 */
include <ESP8266WiFi.h>
#include <Modbus.h>
#include <ModbusIP_ESP8266.h>

//Modbus Registers Offsets (0-9999)
const int SENSOR_IREG = 100;
const int temp_ISTS = 101;            //
const int out11_COIL = 104;           //  these are regs? I figured this was reg numbers right?
const int out22_COIL = 105;           //  and wasn't sure how this works
const int out33_COIL = 106;           //  I was hoping you could help me here, tell me how,or
const int out44_COIL = 107;           //  fix it and send it back to me (jcobern  at  gmx  dot  com)
const int in55_COIL = 108;            //  this compiles, but that dosen't mean it work like it should.
const int in66_COIL = 109;            //  also can you tell me the modbus address I will find these valves
const int in77_COIL = 110;            //   any help you can give will be greatly appreciated 
const int in88_COIL = 111;            //
const int sensorValue = analogRead(A0);  //  in value
const int tempPin = 16;                   //  temp
const int out11Pin = 12;                 // out on-off
const int out22Pin = 13;                 // out on-off
const int out33Pin = 14;                 // out on-off
const int out44Pin = 6;                  // out on-off
const int in55Pin = 7;                   // in on-off
const int in66Pin = 8;                   // in on-off
const int in77Pin = 9;                   // in on-off
const int in88Pin = 10;                  // in on-off
long ts;

//ModbusIP object
ModbusIP mb;

void setup() {

    Serial.begin(115200);
 
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(1500);
    Serial.print(".");
  }

  Serial.print(WiFi.localIP());

  ts = millis();
    //Config Modbus IP
    mb.config("routername", "pw");                 // it connect just fine, then show me the ip
    
    //Set ledPin mode
    pinMode(tempPin, INPUT);
    pinMode(out11Pin, OUTPUT);
    pinMode(out22Pin, OUTPUT);
    pinMode(out33Pin, OUTPUT);
    pinMode(out44Pin, OUTPUT);
    pinMode(in55Pin, INPUT);
    pinMode(in66Pin, INPUT);
    pinMode(in77Pin, INPUT);
    pinMode(in88Pin, INPUT);
    
    
    // Add  register - Use addIsts() for digital inputs
    mb.addIsts(temp_ISTS);
    mb.addCoil(out11_COIL);
    mb.addCoil(out22_COIL);
    mb.addCoil(out33_COIL);
    mb.addCoil(out44_COIL);
    mb.addCoil(in55_COIL);
    mb.addCoil(in66_COIL);
    mb.addCoil(in77_COIL);
    mb.addCoil(in88_COIL);
}

void loop() {
    //Call once inside loop()
   mb.task();
   if (millis() > ts + 2000) {
       ts = millis();
       
   //Attach Pins to registers
   mb.Ireg(SENSOR_IREG, analogRead(A0));   
   mb.Ists(temp_ISTS, analogRead(tempPin));
   mb.Ists(out11_COIL, out11Pin);
   mb.Ists(out22_COIL, out22Pin); 
   mb.Ists(out33_COIL, out33Pin);
   mb.Ists(out44_COIL, sensorValue);
   mb.Ists(in55_COIL, in55Pin);
   mb.Ists(in66_COIL, in66Pin);;
   mb.Ists(in77_COIL, in77Pin); 
   mb.Ists(in88_COIL, in88Pin); 
   }
   
Serial.println(tempPin);
   Serial.println(sensorValue /  83);
   delay(2000);
     // prints title with ending line break   
//this print and delay stuff will be removed after testing

  }

ESP-O-MAT, 2020-04-14:
You add coils (mb.addCoil) and then use mb.Ists. You should use mb.Coil.
For example: mb.Coil(out11_COIL, out11Pin);

Or you can use addIsts in setup:
mb.addIsts(out11_COIL);