andresarmento/modbus-esp8266 #9: Tcp Keep Alive - modbus-esp8266

guiant32, 2016-09-16:
Hi all!

Is there a way to keep tcp alive in modbus-esp8266 like "https://github.com/andresarmento/modbus-arduino" ?

 I commented "client.stop();" in the file "ModbusIP_ESP8266.cpp" but there was no result.

Thanks.

kg9316, 2016-10-05:
Did you find an answer guiant32 ?
I have the same problem.

Quarian3ngineer, 2016-10-19:
Is this topic still active?

toto99303, 2017-04-20:
I have the same question, anyone got keep-alive working?

circuitjaya, 2017-04-22:
The following implementation allows to do multiple reads/write with open TCP connection. 
https://github.com/JhonControl/ESP8266_Industrial_ModbusTCP_V2
Hint: The implementation seems to be is unstable when there is a established tcp connection to ESP8266 and second Modbus TCP master is trying to open the port!.

toto99303, 2017-05-12:
Hi, thanks for your reply. I’ve tried the library you suggested and it’s working fine, but I ran into another problem when the TCP connection is kept open – my HTTP server is not responding…

I will revert to your library, because this way I can have more masters reading from the slave ESP and HTTP will be working.

 

But how to read from SCADA software (Citect) and force the scada software to reconnect after each read? Do you have experience with that?

I’m reading successfully, but only one time and then it doesn’t reconnect…

 

Thanks in advance!

 

Regards,

Antonio Velkov

 

https://github.com/JhonControl/ESP8266_Industrial_ModbusTCP_V2

circuitjaya, 2017-06-05:
@toto99303 : sorry for late reply. Do you still have an issue?

circuitjaya, 2017-06-05:
So far, we could not managed to implement a stable server on ESP that keeps the connection alive due to stack limitations.
As a interim solution, we would suggest to write a simple gateway/bridge application (could run on same PC that runs SCADA).
On one end acting as master/client to  handle open-close for every reading and on another end acting as slave/server to SCADA system (always open connection).

toto99303, 2017-06-05:
I decided not to use modbus for my project. It's still not that stable as I wished...
thanks!

emelianov, 2017-08-19:
Guys, in case if you still interesting in ModBus on ESP i've forked the library and trying to continue support.
Now I implemented keep alive connection, multiple connections at the same time, callback support and ESP32 support.
https://github.com/emelianov/modbus-esp8266

André, if you are reading this, you've done really great job. Thanks!